bitcraze/aideck-gap8-examples #13: relay char delivery from GAP8 through NINA

knmcguire, 2020-06-10:
The UART1 connection will be conflicting with the lighthouse deck. Moreover, the UART connection with the GAP8 will be disrupted when the cluster is turned on, which might cause problems for users in the end anyway.

SuperN1ck, 2020-09-18:
Hey @knmcguire 
are there any updates on this?

We were just discussing within our group whether it would be an alternative to modify the lighthouse deck to use UART2 instead of UART1? 
We guess that would imply some physical modifications on the lighthouse deck (aka re-soldering?) but could also mean less (deep) modifications in the code base?

Do you think it would be possible? What is your opinion on this idea?

knmcguire, 2020-09-18:
Currently I think we are busy with finalizing some firmware stuff on the lighthouse deck and I believe that using a different port that is also connected with the crazyflie. I will get back to you on that once I discussed it with my colleagues when I'm back next week.

RavenLite, 2021-04-20:
Hi, I'm curious why the last column is different from the last row in the table "Deck-deck", they are both about AI-deck.
https://www.bitcraze.io/documentation/system/platform/cf2-expansiondecks/

knmcguire, 2021-04-20:
Ahh good point. We will fix that soon! thanks for notifying.

For now you can just assume that the bottom row counts

neoamos, 2021-05-28:
I implemented the NINA firmware to do this, as well as an interface for the Gap8 and Crazyflie to send and receive packets.
https://github.com/neoamos/Lighthouse-and-AI-Deck-Compatibility

knmcguire, 2021-05-31:
ah this is awesome! I'll make sure to forward this repo to those that have the same limitation.

There are future plans to move the lighthouse connection to CF2 to i2c so hopefully this loop will not be necessary, but currently we don't have a projection of when this will happen.

knmcguire, 2022-03-09:
The new pipeline of the AIdeck gap8 nina handling does exacty this. 1ff1da5795de8f1feff4c97282bcfd07b6bcd3ae

closed 2022-03-09 by knmcguire