bitcraze/aideck-gap8-examples #15: handle raw image to color image

knmcguire, 2020-07-10:
- [ ] Write a simple demosaicking algorithm

- [ ] Stream the color image instead of raw

knmcguire, 2020-07-10:
quite a lot of articfacts still in the image. Need a more sophisticated demosaicking algorithm

noahzn, 2021-11-23:
Hi, could you give an example to stream the color image instead of the raw image? Thanks.

knmcguire, 2021-11-25:
Hi, we don't have this functionality/example yet, and also since the Aideck 1.1 has switched to a grayscale camera it's not our immediate priority. So let me discuss with the team if this is something that we will provide in the coming weeks.

noahzn, 2021-11-26:
Hi, thanks for your reply.
We try to solve this problem but meet some problems. 
https://forum.bitcraze.io/viewtopic.php?t=5150

aqqz, 2021-12-18:
Hello, I have a question, why the color of the output rgb image is so green, can I adjust it？

knmcguire, 2022-01-12:
an update here: please check out this readme to get the color image demosaicking on the computer example https://github.com/bitcraze/AIdeck_examples/tree/cf-link/GAP8/test_functionalities/wifi_jpeg_streamer

It's still a bit messy but we hope to clean it out for the next couple of days.

closed 2022-03-09 by knmcguire