bitcraze/aideck-gap8-examples #16: ai deck face detection

4perros, 2020-07-22:
hello
i am using ubuntu 18.04 and have gap_sdk version 3.6 my issue is that when i run this example i keep getting this error Makefile:48: recipe for target 'GenTile' failed 
make: *** [GenTile] Error 1.  Could you let me know how to solve this problem. 
thank you

knmcguire, 2020-07-22:
Hi! Thank you for letting us know. The face detection example was last tested on the SDK version 3.4. I indeed got the same error as you did on 3.5 already. I also have 18.04 and after [downgrading the sdk to 3.4](https://github.com/GreenWaves-Technologies/gap_sdk#upgradingdowngrading-the-sdk), (and `make gap_tools`), it compiled for me.

I will update the documentation to state this, since the SDK version was not displayed there. In the mean time we will look into updating the example to work for 3.6

knmcguire, 2020-07-22:
Full error message from my side:

```
gcc -o GenTile -I/home/kimberly/projects/greenwave/gap_sdk/tools/autotiler_v3/include FaceDetGenerator.c FaceDetModel.c /home/kimberly/projects/greenwave/gap_sdk/tools/autotiler_v3/lib/libtile.a -O3 -g -D__PMSIS__ -DUSE_CAMERA -DUSE_STREAMER -DNB_FRAMES=-1 -DHIMAX
FaceDetGenerator.c: In function ‘GenerateResize’:
FaceDetGenerator.c:95:38: error: ‘LOC_INNER_LOOP’ undeclared (first use in this function); did you mean ‘LOC_LOOP’?
   Calls(1, Call("KerResizeBilinear", LOC_INNER_LOOP,
                                      ^~~~~~~~~~~~~~
                                      LOC_LOOP
FaceDetGenerator.c:95:38: note: each undeclared identifier is reported only once for each function it appears in
FaceDetGenerator.c: In function ‘GenerateIntegralImage’:
FaceDetGenerator.c:126:34: error: ‘LOC_INNER_LOOP_PROLOG’ undeclared (first use in this function); did you mean ‘LOC_LOOP_PROLOG’?
    Call("KerIntegralImagePrime", LOC_INNER_LOOP_PROLOG,
                                  ^~~~~~~~~~~~~~~~~~~~~
                                  LOC_LOOP_PROLOG
FaceDetGenerator.c:132:36: error: ‘LOC_INNER_LOOP’ undeclared (first use in this function); did you mean ‘LOC_LOOP’?
    Call("KerIntegralImageProcess", LOC_INNER_LOOP,
                                    ^~~~~~~~~~~~~~
                                    LOC_LOOP
FaceDetGenerator.c: In function ‘GenerateSquaredIntegralImage’:
FaceDetGenerator.c:165:34: error: ‘LOC_INNER_LOOP_PROLOG’ undeclared (first use in this function); did you mean ‘LOC_LOOP_PROLOG’?
    Call("KerIntegralImagePrime", LOC_INNER_LOOP_PROLOG,
                                  ^~~~~~~~~~~~~~~~~~~~~
                                  LOC_LOOP_PROLOG
FaceDetGenerator.c:171:43: error: ‘LOC_INNER_LOOP’ undeclared (first use in this function); did you mean ‘LOC_LOOP’?
    Call("KerSquaredIntegralImageProcess", LOC_INNER_LOOP,
                                           ^~~~~~~~~~~~~~
                                           LOC_LOOP
FaceDetGenerator.c: In function ‘GenerateCascadeClassifier’:
FaceDetGenerator.c:211:31: error: ‘LOC_INNER_LOOP’ undeclared (first use in this function); did you mean ‘LOC_LOOP’?
    Call("KerEvaluateCascade", LOC_INNER_LOOP,
                               ^~~~~~~~~~~~~~
                               LOC_LOOP
Makefile:48: recipe for target 'GenTile' failed
```

knmcguire, 2020-07-23:
My latest commit should solve your problem. I couldn't test it on 3.6 since I got some build issues on that version, so this was tested on 3.5, but I hope it works on 3.6

knmcguire, 2020-07-28:
Could you give us an update if this works for you?

4perros, 2020-07-31:
hello i am still having issues, but i think it is due to the compiler i am using (espressif esp 32) i followed your advice and downgraded to version 3.5 and reinstalled the examples folder and the previous problem dissappeared. i think we can close this issue. i will let you know how i am progressing. thank you for your help.

knmcguire, 2020-07-31:
Ah oke good!

 Is the espressif compiler for the NINA? You could open up a new issue about that if you want, but we will close this one.

closed 2020-07-31 by knmcguire