bitcraze/aideck-gap8-examples #20: Missing Demosaicking Function

SuperN1ck, 2020-08-27:
Hey @knmcguire ,

it seems like in this commit: https://github.com/bitcraze/AIdeck_examples/commit/6151a481554de18c2a3b5b5606f3053ed22bfba3
you moved the demosaicking-function to a seperate file (something like `common/img_proc.c/h`?) but the file was not commited. Is it possible to commit the demosaicking-function or was it left out on purpose?

This also breaks the build of the test_camera example: https://github.com/bitcraze/AIdeck_examples/blob/0517b2b9c1924c052ac42ec533d69d8f4094e57c/GAP8/test_functionalities/test_camera/test.c#L23

Best,
-Nick

knmcguire, 2020-08-31:
Sorry for the late reply but yes you are absolutely right. Commiting it now.

knmcguire, 2020-08-31:
I will close this ticket for now. Would be good to continue on #15 if you think there is something wrong with it. I must warn that demosaicking function does slow down everything though. We might want to parallize it and put it on the cluster.

closed 2020-08-31 by knmcguire

SuperN1ck, 2020-08-31:
Thank you! :+1:  And yes you are right, I somewhat forgot the other issue, my mistake.