bitcraze/aideck-gap8-examples #22: Standalone streamer demo / NINA flashing documentation

gemenerik, 2020-09-07:
Recently got an AI-deck. 

* Can the AI-deck be used as standalone without a Crazyflie by powering via VCOM with 3.7V? If yes, could not see a WiFi-AP when powering it standalone.
* While flashing the NINA, the JTAG one-liner mentioned in the readme of the NINA directory doesn't work out of the box. Consider expanding the documentation a bit.

Had to do the following before it could flash:
1. Add `adapter_khz 20000` line to the `olimex-arm-usb-tiny-h.cfg`
2.  Copy the `tcl` folder from `openocd-esp32` to the `NINA/firmware` dir
3. Run the 'make flash' command in the 'NINA/firmware dir'
3.  Rename `program_esp32` in the one-liner to `program_esp`:
```
~/esp/openocd-esp32/src/openocd -f interface/ftdi/olimex-arm-usb-tiny-h.cfg -f board/esp-wroom-32.cfg -c "program_esp build/partitions_singleapp.bin 0x8000 verify" -c "program_esp build/bootloader/bootloader.bin 0x1000 verify" -c "program_esp build/ai-deck-jpeg-streamer-demo.bin 0x10000 verify reset exit"
```

knmcguire, 2020-09-07:
Hi! 

About your questions:
- Yes this is supposed to work, so that is strange that you can not get the Nina to turn on. We will try to recreate your problem.
- Ah interesting. You did install the esp sdk version 3.3.1 like instructed in the [build documentation](https://github.com/bitcraze/AIdeck_examples/blob/master/docs/nina-instructions/nina-flash-instructions.md)? The exact instructions do work when the docker image is used (found [here](https://github.com/bitcraze/AIdeck_examples/tree/master/docker/nina).

gemenerik, 2020-09-08:
Thanks for your reply!

* If it is supposed to work, I'll have another go and confirm it is powered and flashed correctly.

* All I could think of is that you might use a different OpenOCD version. The version is not specified in the docs, so I used the latest. 

To flash NINA and GAP8 I immediately went for the flashing instructions in the docs and thus completely overlooked the docker images, maybe mention they exist in the documentation.

gemenerik, 2020-09-08:
It probably wasn't flashed correctly. The streamer demo works standalone now.

closed 2020-09-08 by gemenerik

knmcguire, 2020-09-09:
It could be that you got your AIdeck from another source, that hasn't been preflashed. Happy it works now!

The docker images were not quite ready yet and still in progress but we will put them in the documentation soon.