bitcraze/aideck-gap8-examples #24: ai deck

4perros, 2020-09-22:
hello 
i have uploaded the face detection image on the ai deck, now i need to know how to run this code and receive images on the laptop. Could you explain how i can achieve this. Do i need to build an application to run on the laptop? 
thank you
regards
rodrigo calvo

knmcguire, 2020-09-22:
If you make the face detection example with the define USE_STREAMER, then you should be able to stream the camera images to your computer. Just a warning though, the streamer does not work in SDK 3.6 or 3.7

Then check out the documentation for the wifistreamer: https://github.com/bitcraze/AIdeck_examples/blob/master/docs/test-functions/wifi-streamer.md

knmcguire, 2020-09-22:
btw, we also have a AIdeck group on our forum: forum.bitcraze.io. That is a better location for questions that are not bug fixes or feature request.

knmcguire, 2020-09-28:
I've added more info to the documentation so I will close this issue now

closed 2020-09-28 by knmcguire