bitcraze/aideck-gap8-examples #28: Is there any advice for flashing the nina module using the docker on the macOS?

RavenLite, 2020-10-02:
Hello, I firstly used the docker to flash the nina under the official ubuntu virtual machine, but the command didn't work as expected. There is something trouble of the command 'cd' in the dockerflie.
So, I turn to use the docker under the local environment, macOS.
But I meet an another problem when I want to execute the 'docker run' command, 
```
docker run --rm -it -v $PWD:/module/ --device /dev/ttyUSB0 --privileged -P espidf:3.3.1 /bin/bash -c "make menuconfig; make clean; make all; /openocd-esp32/bin/openocd -f interface/ftdi/olimex-arm-usb-tiny-h.cfg -f board/esp-wroom-32.cfg -c 'program_esp32 build/partitions_singleapp.bin 0x8000 verify' -c 'program_esp32 build/bootloader/bootloader.bin 0x1000 verify' -c 'program_esp32 build/ai-deck-jpeg-streamer-demo.bin 0x10000 verify reset exit'"
```
You know, there is not device file in the macOS, so, I can't set the value of the `--device`, is there any advice to solve it?
Thanks!
> I finish the work according to the [Docker Nina](https://github.com/bitcraze/AIdeck_examples/blob/master/docs/nina-instructions/docker-nina.md)

RavenLite, 2020-10-04:
After some research, I think maybe it's a better choice to back to the ubuntu virtual machine.
But I met another problem
When I run the `docker run` command,
It says that "Error: unable to open ftdi device with vid..."
***
Let me to solve the problem.

RavenLite, 2020-10-04:
[image]

knmcguire, 2020-10-05:
Currently there is a forum thread open for the VM problems.

But here let's continue to talk about getting this to work on Mac, since that is where the original issue was about. @krichardsson might be able to have some insight on this, but he was not sure if this is possible on Mac.

krichardsson, 2020-10-05:
Hi @RavenLite !
I was looking a bit at the problem of accessing devices from docker on OSX. There is no simple solution but I found this article https://christopherjmcclellan.wordpress.com/2019/04/21/using-usb-with-docker-for-mac/ that describes the same problem and a solution. It is not straight forward but probably works. The solution is to replace the virtual machine that comes with docker (behind the scenes) on Mac with Virtual Box. I have not had the time to try it out.

RavenLite, 2020-10-07:
Hi! Thanks for your help~
I might find the problem, it's that I have been flashing using the Crazyradio PA.
But the @kimberly in the "Bitcraze Forum" told me that 
> "Crazyradio PA is not used to program the Nina. You need a compatible JTAG programmer/debugger in order to program the Nina, just like the GAP8."  [Source](https://forum.bitcraze.io/viewtopic.php?f=21&t=4473)

I think maybe the introduction doc should mention it?
I'm not familiar with the embedded programming before, so maybe there is some knowledge that you think users have by default. But others don't know?
Let me help to optimize the doc in the view of an embedded programming freshman.

knmcguire, 2020-10-07:
The fact that you need a jtag based programmer for the aideck is in bold on our [store page](https://store.bitcraze.io/collections/decks/products/ai-deck), [getting started guide](https://github.com/bitcraze/AIdeck_examples/blob/master/docs/getting-started/getting-started.md) and on the [release blog](https://www.bitcraze.io/2020/06/ai-deck-is-available-in-early-access/). But if you bought it at a different store, maybe this information was not noted there?

knmcguire, 2020-10-07:
But for an embedded programmer freshman, you can already gain some experience by trying things out on the crazyflie firmware itself. It's easier to start out with than the gap8 (since it is an STM32f4), and [firmware repo](https://github.com/bitcraze/crazyflie-firmware) has some examples of the [ app layer](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/userguides/app_layer/) also. Might be good to start out with those and work your way up.

RavenLite, 2020-10-07:
Yes, I read every related docs about AI-deck above.
But I don't realize that I can't flash the NINA firmware using Crazyradio PA~
Thanks for your learning suggestion!!

closed 2020-10-08 by RavenLite