bitcraze/aideck-gap8-examples #29: problem uploading to ai deck

4perros, 2020-10-06:
Hello kimberley
I installed the gap v3.7, I run the test_camera example and it worked and I was surprised to get a color picture, then I  installed the face detection example, when I run it I couldn’t get it to work, I changed the setting in the makefile to USE_CAMERA, but it doesn’t work, so I tried to run the test camera example again, now I keep getting this message 
Info: gdb port disabled
Info: halt timed out, wake up GDB
error:  timed out while waiting for target halted
i can not flash any other example
I noticed that only the NINA led is coming on. 
Could you let me know what I can do to fix this problem
regards
rodrigo

knmcguire, 2020-10-06:
Hi,

I haven't see this problem before. It might be that you need to reconnect and connect your programmer. In the worst case maybe you should restart the computer (kind of clear out all USB permissions or other kind of settings.). 

What kind of programmer are you using?

4perros, 2020-10-06:
thank for your reply i am using the espressif/esp-idf programmer. it has two buttons one for boot and and one for reset, i have tried with this buttons but nothing changes, i will do what you said and see what happens. thank you again
rodrigo

4perros, 2020-10-06:
this is the programmer i am using

[image]

knmcguire, 2020-10-06:
ah oke. We usually use the olimex programmer given on this page:

https://github.com/bitcraze/AIdeck_examples/blob/master/docs/getting-started/jtag-programming.md

We also have the esp programmer that you use, but it does not work as many times as with the olimex (although with that one we also have to occasionally unplug and replug the programmer).

We will try it again with the esp programmer one of these days.

knmcguire, 2020-10-07:
Just an update.

I tried the esp-prog and managed to flash the gap8 with SDK 3.5 (the streamer func is broken on 3.7, see this [issue](https://github.com/bitcraze/AIdeck_examples/issues/19)). With **make clean all run** I get this issue though:

```
	 *** Wifi Streamer ***

Entering main controller...
Failed to allocate Memory for Image 
Traceback (most recent call last):
  File "/gap_sdk/tools/gapy/gapy", line 162, in <module>
    main()
  File "/gap_sdk/tools/gapy/gapy", line 152, in main
    operationFunc(args, config)
  File "/gap_sdk/tools/gapy/run.py", line 146, in operationFunc
    raise RuntimeError('Runner has failed with value: %d' % status)
RuntimeError: Runner has failed with value: -1
/gap_sdk/tools/rules/pulp_rules.mk:213: recipe for target 'run' failed
make: *** [run] Error 1
```

but the **make clean all image flash** works. 

I did however, had to do the following procedure everytime:

1. Disconnect esp-prog from both computer and aideck
2. Connect esp-prog to computer
3. Connect esp-prog to gap8 jtag connector
4. Turn on crazyflie

Not sure if this order makes sense... but at least it worked for that. It's not handy at all though so maybe we should remove this programmer from the list at all. 

I tried the esp-prog with the Nina and that worked just fine. I think that the fact that the esp-prog doesnt handle the reset correctly is something that the gap8 chip does not like (since it is necessary to program the chip)

4perros, 2020-10-08:
hello
i did try your suggestion but i am still unable to run any examples. I am going to purchase the Olimex programmer instead, and i think it will be a good idea not to endorse that product, that's what i think.
I will let you know how i go when i get new programmer, and thank you again for your advise.
regards
rodrigo

knmcguire, 2020-10-08:
Probably for the best. Did you buy the esp-prog because of the list or did you already had in your possession? If it was the first we are sorry it didn't work out. 

It's probably the smartest to use the same programmer as we do. Let us know if you have anymore problems

RavenLite, 2020-10-09:
Hi, knmcguire
I see that you give a buying link in the [JTAG adapter/debugger](https://github.com/bitcraze/AIdeck_examples/blob/master/docs/getting-started/jtag-programming.md) for the [Olimex ARM-JTAG-20-10](https://www.olimex.com/Products/ARM/JTAG/ARM-JTAG-20-10), one of two necessary extra hardwares for flashing the nina or gap8.
Could you please give a buying link for the other one, Olimex ARM-USB-TINY-H, maybe [this one](https://www.amazon.com/Olimex-ARM-USB-TINY-H-Arm-Jtag/dp/B009UED630)?

In fact, I bought an another cheaper JTAG debugger but I can't find the cfg file for that one, used in the docker command in the [Docker Nina](https://github.com/bitcraze/AIdeck_examples/blob/master/docs/nina-instructions/docker-nina.md)
```
/openocd-esp32/bin/openocd -f interface/ftdi/olimex-arm-usb-tiny-h.cfg
```

What's more, I want to ask that where is the folder `interface`? I don't find it in the AI-deck project.
***
Thank you so much!!!!!!

knmcguire, 2020-10-12:
Hi, you need both the olimex programmer and the jtag adapter. Where you can buy the Olimex ARM-USB-TINY-H depends on which region you are, but I guess digikey would be good:

https://www.digikey.se/product-detail/en/olimex-ltd/ARM-USB-TINY-H/1188-1013-ND/3471388

The interface folder for the gap8 project is in the openocd-esp32 folder, which is part of the docker container you build (not part of the aideck itself). You can open the docker container:

docker run --rm -it espidf:3.3.1 /bin/bash and then navigate to
`cd ../../openocd-esp32`

But maybe its just easier to go to the repo itself: https://github.com/espressif/openocd-esp32. That is what the docker container has cloned exactly in build. You can check in there (tcl/interface) if there is any .cfg file for you programmer. 

If this works for your programmer on both the ESP and gap8, let us know! And let us know which programmer you use ofcourse

RavenLite, 2020-10-12:
Thanks!
I bought Olimex ARM-USB-TINY-H JTAG. It works so fluently! (Smile!~
BTW, I use the docker way to do that. And I find some information worthy to be added to the docs, I will create another PR to do that.

knmcguire, 2020-10-13:
Ah perfect! I'll check your PR but thanks for letting us know. 

I'll close this issue for now.

closed 2020-10-13 by knmcguire

tobbeanton, 2020-10-22:
I think one problem with the ESP-Prog is that it does not wire the reset to the JTAG interface, only to the PROG interface. It is pin nr 10 in the picture. I don't know why they left this unconnected but probably because the EPS32 doesn't handle reset that well anyway and doesn't reset to a defined state were JTAG can be attached to. 
[image]

It should be possible to use a soft reset but if the CPU is locked early after reset in a state were it is unresponsive, it might not be possible to attach to it.

Too bad since the EPS-Prog offers a FT2232H JTAG interface with the right cable.