bitcraze/aideck-gap8-examples #30: nntool not working for gap_sdk version 3.7 anymore

SuperN1ck, 2020-11-01:
Hey all,

in the release version `3.7` of the gap sdk there is a hard-coded download from a master branch of tensorflow, see here: https://github.com/GreenWaves-Technologies/gap_sdk/blob/d33d6c90c6a6d1e87572515f61ee8642e09847f0/tools/nntool/Makefile#L56

Unfortunately, this breaks the nntool for version `3.7` when using the docker. A quick workaround that worked for me is to replace `master` by some older commit like `6be604aaacd9d270de01c37ec6e9a9a077397848` being inside the docker then rebuilding the nntool and commiting my changes. But I guess that is not substantially enough for everybody.

FYI, in the release version `3.7.2` of the gap sdk this was fixed (https://github.com/GreenWaves-Technologies/gap_sdk/blob/a934747441481ea3d9c029719d721780cdff9e46/tools/nntool/Makefile#L58) but building the docker with version `3.7.2` results in a different issue which was already addressed here: https://github.com/GreenWaves-Technologies/gap_sdk/issues/174

Best,
-Nick

knmcguire, 2020-11-03:
hmmm... that's not great. Is this more an issue for the SDK though you think?

knmcguire, 2020-12-04:
btw, how about the new SDK ? There is a updated docker file in #35 as well

SuperN1ck, 2020-12-13:
Hey,
sorry for the late reply. I can confirm that with version 3.8.1 the nntool is working again.

closed 2020-12-13 by SuperN1ck

knmcguire, 2020-12-13:
great! Thanks for letting us know