bitcraze/aideck-gap8-examples #31: Cannot flash any example using Olimex ARM-USB-TINY-H

Gualor, 2020-11-14:
Hi,
I'm working on a project on the AIdeck which only implements GAP8 chip + NINA module functionalities. I bought an Olimex ARM-USB-TINY-H with ARM JTAG 20 to 10 pin adapter, as suggested, to program the board.

Saddly I couldn't make it to work, both by building the GAP_SDK myself and by using the docker image provided.

I will post the output I receive from every example of GAP8 i tried:

```sh
Open On-Chip Debugger 0.10.0+dev-00002-ga9347474 (2020-11-14-17:27)
Licensed under GNU GPL v2
For bug reports, read
  http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: gap8.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
Warn : Burst read timed out
```

I'm using the latest release of gap_sdk on an Ubuntu 20.04 machine.
(note: out of the box, when powered, AIdeck setup the WI-FI hotspot and stream the camera images as expected)

Thank you in advance for your time.

Best regards,

Lorenzo Gualniera

Gualor, 2020-11-14:
Update: I managed to solve the issue by swapping a few cable connections between Olimex ARM JTAG TINY H and the 20 to 10 pin adapter, I just supposed the 20-10 connector was mirrored (as other products are). So this part of the issue was not due to a bug on the AIdeck side.

However, after the JTAG programming was successful the example I was testing `wifi_jpeg_streamer` gave the following error:
```sh
test.c:4:10: fatal error: tools/frame_streamer.h: No such file or directory
 #include "tools/frame_streamer.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

knmcguire, 2020-11-16:
Have you build the SDK with 'make gap_tools' and 'make pulp_os' ? Btw, you should use SDK version 3.5 for the wifi streamer example since we are still experiencing a bug with 3.6 and up.

Gualor, 2020-11-16:
Hi,
thank you for the reply, I followed the full installation of the tutorial (which I assume also installed pulp_os and gap_tools) using the command `make sdk`. Thank you for the suggestion, I will try out `gap_sdk v3.5` as soon as I can.

Another question, I saw in the example that NINA uses freeRTOS while GAP8 uses PULP OS, is there any particular reason for this choice? Would it be possibile to use freeRTOS on both MCU?

Best,
Lorenzo

tobbeanton, 2020-11-16:
> Update: I managed to solve the issue by swapping a few cable connections between Olimex ARM JTAG TINY H and the 20 to 10 pin adapter, I just supposed the 20-10 connector was mirrored (as other products are). So this part of the issue was not due to a bug on the AIdeck side.
> 
> However, after the JTAG programming was successful the example I was testing `wifi_jpeg_streamer` gave the following error:
> 
> ```shell
> test.c:4:10: fatal error: tools/frame_streamer.h: No such file or directory
>  #include "tools/frame_streamer.h"
>           ^~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> ```
Could you elaborate the solution a bit more? We did not see this problem so maybe something has changed with their 20-10 connector?

Gualor, 2020-11-16:
Yes I will elaborate my previous answer. As it was the first time using this JTAG programmer I wasn't familiar with the pinout, I have both an Olimex arm usb tiny h and Olimex arm jtag 20-10 pin connector, both of them have female pin headers. 

The beginner mistake I made was assuming the pinout of the two devices were mirrored so by using jumper cables I just connected 1:1 the pins (in the same position but flipped), but in reality the pinout is exactly the same (not the reflection) and so the jumper connections made are now not as tidy (had to swap and twist a row of pins), but now it works perfectly.

Couldn't find any images of the two devices connected together, I hope my explanation was sufficiently clear.

Best,
Lorenzo

tobbeanton, 2020-11-16:
Thanks for the clarification. We use a 1.27mm flat cable like in [this](https://github.com/bitcraze/AIdeck_examples/blob/master/docs/getting-started/jtag-programming.md) picture but we have also experienced that the Olimex flat cable can be a bit "glitch".

closed 2020-11-27 by Gualor