bitcraze/aideck-gap8-examples #39: HyperFlash and HyperRAM

SuperN1ck, 2020-12-23:
Hey,
I'm kinda confused regarding memory/RAM on the AI-Deck. On the product/github page the AI-Deck is advertised to have 
512 Mbit HyperFlash and 64 Mbit HyperRAM but when I look at the [GAP8 product page](https://greenwaves-technologies.com/gap8_gap9/)
It's much less. Is this a typo? But, when I look at the [schematics](https://www.bitcraze.io/documentation/hardware/ai_deck/ai-deck-revc.pdf) of the AI-Deck it's also mentioned there.
So, if this hypermemory exists is there any example code showing how to use the extended memory?

Thanks,
-Nick

knmcguire, 2020-12-23:
Hi! it seems confusing indeed but i'm sure that is a explanation. I will nudge the GW team for this but it might be difficult to get an answer before Christmas, so I hope this is not in a hurry?

Anyway happy holidays!

Yaooooo, 2021-01-05:
Hi @SuperN1ck 

There's a confusion between internal L2 RAM (512KB), and external Memory on HyperBUS. 
On GAP8, we have 512KB L2 and 64KB L1 memory embeded, without Flash.
Thanks to Hyperbus and Quad-SPI interfaces, we can connect to an external RAM and an external Flash.
For example, on AI Deck, it's equipped with an external RAM + Flash, called HyperFlash and HyperRam.
You can find some related examples from our SDK: 

- HyperFlash: https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/pmsis/bsp/flash/hyper_flash
- HyperRam: https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/pmsis/bsp/ram/hyper_ram

I see also one test from AI Deck repo: https://github.com/bitcraze/AIdeck_examples/tree/master/test/gap8_test

Best

SuperN1ck, 2021-01-14:
Thank you for the answer!

Do you know by any chance if it is possible to use the HyperRam similar to the L1/L2 memory. For instance such that I can do something like
```
int* test = pmsis_l3_malloc(sizeof(int) * 5)
```
Or do I always have to explicitly copy the memory through a task?

Best,
-Nick

Yaooooo, 2021-01-14:
No, since the L3 Memory (external memory) is not memory mapped in our system, and we don't have MMU, we cannot use it as L2/L1 to have the direct access. 
It's possible to have such an api like "pmsis_l3_malloc", but since it's an external device, which is managed in the BSP level. However, allocator is in runtime level. We won't mix them, since it will cause maintain issues in the future. 
Do you see any problem in using our API for managing the external RAM ?

Yao

SuperN1ck, 2021-01-15:
Thank you for your info! Luckily I found a way such that our program fits in L2 only (for now). I will close that issue now since my initial question was answered.

So the following is more a guess not concrete feedback:
There is no problem with the API itself. But as far as I understand it's not possible to use the whole L3 memory at the same time since I always need to copy the parts I need to the L2 in order to do computations. In that case the program is then limited by the L2 memory size? If I want to process more than in the L2 fits, I have to copy the first chunk from L3 to L2, process it, copy back from L2 to L3, then the second chunk from L3 to L2 etc I guess. I'm not sure but maybe the Autotiler takes care of that?
Happy to your thoughts :)

closed 2021-01-15 by SuperN1ck

Yaooooo, 2021-01-15:
Exactly, that's why we developed Autotiler.
You can take a look into our website:
https://greenwaves-technologies.com/tools-and-software/

Especially the GAPFlow and Autotiler.
You are welcome to ask any questions :)

Yao Zhang
Greenwaves Technologies

huzq85, 2022-11-18:
> Hi @SuperN1ck
>
> There's a confusion between internal L2 RAM (512KB), and external Memory on HyperBUS. On GAP8, we have 512KB L2 and 64KB L1 memory embeded, without Flash. Thanks to Hyperbus and [...]

hi, Yong,
Thank you for your comments but when I am trying to dive into the examples you shared, it looks like the links are invalid. Do you have the latest examples of how to allocate the HyperRAM that I can learn from? Thanks!

huzq85, 2022-11-18:
> Sorry, the sdk has been reorged... Please find the path here:
> 
> * Hyperflash: https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/gap8/basic/bsp/flash
> * Hyperram: https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/gap8/basic/bsp/ram
> 
> Yao

hi, Yao,
Thank you for your real-time response! I will take a look at these examples first and get back to you if I have questions. I appreciate your help!