bitcraze/aideck-gap8-examples #40: wifi_streaming example returns half-corrupted images

LorenzoLamberti94, 2021-01-14:
Hi!
I am working on the AIdeck and I am trying to make the wifi streaming example work.

I am using the last bitcraze commit and last gap_sdk version 3.8.1.

My problem is that when I run the viewer.py script to visualize the streamed images, the bottom part of the image is either out of sinc or black.
I debugged hardly the greenwaves code of the streamer, but I didnt find the problem.

On AIdeck side: after the camera acquisition I dump the raw image, this looks correct. So the camera works.
If I run the test camera program (write the image to the PC via debug interface) I always get the right image. This proofs again the working camera.

The problem I think is either in the Jpeg encoder, in the trasmission, or its a buffer size problem at receiving side.

in the viewer.py file i tried many buffer sizes (wrt to the original value of 512). I went down to 100 bytes and up to 10000 bytes, but the same problem occurs.

does anyone has the same problm that I do?
refer here for an example image: https://forum.bitcraze.io/viewtopic.php?f=21&t=4643

knmcguire, 2021-01-19:
Hi, Lets talk further here then since this is an actual bug.

There is probably something wrong indeed. Unfortunately the JPEG compression written in the SDK is not conventional. You mentioned the double buffering before in the forum, but not sure if that will fix the problem either. 

This week we will not have time to look in this in more detail so this needs to be done next week at our side.

LorenzoLamberti94, 2021-01-22:
Thank you kimberly

I printed the stream of data on both sender (GA8 to NINA) and receiver side (laptop Python script)

i might have found the source of the problem: the last packet sent is sent after the end of image string (footer = 0xffd9). 
To better unverstand, this is what we should have:
header
packet
....
packet
packet
footer

what we are actually receiving
header
packet
....
packet
footer
packet

It seems also that GAP8 communicates only once to NINA the header and footer of JPEG, and NINA actually attach them to the bitstream. [In this file.](https://github.com/bitcraze/AIdeck_examples/blob/master/NINA/firmware/main/ai-deck-jpeg-streamer-demo.c)

I''ll look further next week. Thanks!

knmcguire, 2021-01-25:
The streamer file on the gap8 side were sending the jpeg in a unconventional way by only sending the header and footer once first, so that is why the NINA is patching this up. But maybe something went wrong in the patching part of things

knmcguire, 2021-03-16:
Hi! Good call!

I will give it a go this week. I will need to go to the office to pick up a programmer since the one that I have here is malfunctioning, so once I have I can try it out.

evoggy, 2021-03-16:
Yes, this seems to be the issue. There was also some changes needed in the python script that receives the image. I've pushed a new version of the Nina application and the python script, could you give it a try and see if it solves the issue?

knmcguire, 2022-03-09:
This has been completely reworked with the new AIdeck example, so this issue will probably no longer hold. If it does, please reopen!

closed 2022-03-09 by knmcguire