bitcraze/aideck-gap8-examples #41: Update to SDK 3.9

knmcguire, 2021-03-22:
The newest sdk 3.9 is out, so we should update the docker files and test out all the examples

bot-motion, 2021-05-05:
I wanted to let you know that I created a SDK 3.9.1 dockerfile, available here:

https://github.com/bot-motion/aideck-dev

you can test it and copy it if you like.

knmcguire, 2021-05-26:
There was a forum post about this recently: https://forum.bitcraze.io/viewtopic.php?f=21&t=4871&p=22251#p22251

@bot-motion thanks for providing it! sorry for the late response. Did you manage to use this docker file on the ai mnist example?

neoamos, 2021-05-28:
I tested it with the simatai/gapsdk:3.9.1 provided by @bot-motion and got the following error
```
Using TensorFlow backend.
Traceback (most recent call last):
  File "model/train.py", line 21, in <module>
    import keras
  File "/usr/local/lib/python3.6/dist-packages/keras/__init__.py", line 3, in <module>
    from . import utils
  File "/usr/local/lib/python3.6/dist-packages/keras/utils/__init__.py", line 6, in <module>
    from . import conv_utils
  File "/usr/local/lib/python3.6/dist-packages/keras/utils/conv_utils.py", line 9, in <module>
    from .. import backend as K
  File "/usr/local/lib/python3.6/dist-packages/keras/backend/__init__.py", line 1, in <module>
    from .load_backend import epsilon
  File "/usr/local/lib/python3.6/dist-packages/keras/backend/load_backend.py", line 90, in <module>
    from .tensorflow_backend import *
  File "/usr/local/lib/python3.6/dist-packages/keras/backend/tensorflow_backend.py", line 5, in <module>
    import tensorflow as tf
ModuleNotFoundError: No module named 'tensorflow'
train_model.mk:18: recipe for target 'model/model.h5' failed
make: *** [model/model.h5] Error 1
```

[Dockerfile.zip](https://github.com/bitcraze/AIdeck_examples/files/6560403/Dockerfile.zip)
This is the docker file that I created and was using.  I basically just commented out the `RUN cd gap_sdk; pip3 install -r tools/nntool/requirements.txt` because it was moved into the make tasks, and it no longer executes correctly.
The AI example works up to the end of the "all" task when it stalls trying to flashes the data for the model on to the the deck.  If you terminate the all task and then run "make run", it begins to execute on the deck, but the constructor for the neural network crashes.

```
	 *** NNTOOL model Example ***

Entering main controller
Reading image
Going to alloc the image buffer!
Image /module/data/samples/5223_5.pgm:  [W: 28, H: 28] Bytes per pixel 1, HeaderSize: 13
P5
28 28
255

Image /module/data/samples/5223_5.pgm, [W: 28, H: 28], Bytes per pixel 1, Size: 784 bytes, Loaded successfully
Constructor
Graph constructor exited with an error
Traceback (most recent call last):
  File "/gap_sdk/tools/gapy/gapy", line 162, in <module>
    main()
  File "/gap_sdk/tools/gapy/gapy", line 152, in main
    operationFunc(args, config)
  File "/gap_sdk/tools/gapy/run.py", line 146, in operationFunc
    raise RuntimeError('Runner has failed with value: %d' % status)
RuntimeError: Runner has failed with value: -1
/gap_sdk/tools/rules/pulp_rules.mk:244: recipe for target 'run' failed
```

bot-motion, 2021-05-29:
Hello, I have not tested the docker image for GAP SDK 3.9.1 for the MNIST example. I don't think it is possible to skip the requirements.txt for the NNTool, as that installs Keras - see

https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/requirements.txt

At the moment I bricked my NINA chip, so I'm far away from AI ;-). The docker image has worked well for me to flash the GAP with most functionality examples, but I have not run the python parts of it yet.

@neoamos I don't understand what you mean by "because it was moved into the make tasks" -- it is true though that tensorflow is conspicously absent in the image:
```
(base) botmotion:~$ bash-gap8
root@gap8:/module# python3
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow as tf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'tensorflow'
>>> import keras as k
Using TensorFlow backend.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/keras/__init__.py", line 3, in <module>
    from . import utils
  File "/usr/local/lib/python3.6/dist-packages/keras/utils/__init__.py", line 6, in <module>
    from . import conv_utils
  File "/usr/local/lib/python3.6/dist-packages/keras/utils/conv_utils.py", line 9, in <module>
    from .. import backend as K
  File "/usr/local/lib/python3.6/dist-packages/keras/backend/__init__.py", line 1, in <module>
    from .load_backend import epsilon
  File "/usr/local/lib/python3.6/dist-packages/keras/backend/load_backend.py", line 90, in <module>
    from .tensorflow_backend import *
  File "/usr/local/lib/python3.6/dist-packages/keras/backend/tensorflow_backend.py", line 5, in <module>
    import tensorflow as tf
ModuleNotFoundError: No module named 'tensorflow'
>>> 

```

neoamos, 2021-05-29:
@bot-motion I mean that the python requirements for nntool was integrated into the make tasks for nntool.
If you look at the Makefile for nntool, the task "req" installs these requirements.  So when you do "make all" for the gap sdk, it will install the python requirements for nntool as well.
https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/Makefile

I installed tensorflow in your docker container, and it had the same behavior as mine.

I looked a bit more into the flash command that is failing.
```
Flashing image with command:
gap8-openocd -c "gdb_port disabled; telnet_port disabled; tcl_port disabled" -c "script interface/jlink.cfg; script target/gap8revb.tcl; script tcl/flash_image.tcl; script tcl/jtag_boot.tcl; gap_flash_raw_hyper /module/data/BUILD/GAP8_V2/GCC_RISCV/target.board.devices.flash.img 284164 /gap_sdk/tools/gap8-openocd-tools; exit;"
Open On-Chip Debugger 0.10.0+dev-dirty (2021-05-07-15:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : J-Link V11 compiled Aug 14 2019 16:21:09
Info : Hardware version: 11.00
Info : VTarget = 2.838 V
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
--------------------------
begining flash session (hyperflash)
--------------------------
load flasher to L2 memory
Loading binary through JTAG
Warn : Burst read timed out
[Flahser]: Hyperflash flasher is ready
Instruct flasher to begin flash per se
device struct address is 470047304
Warn : Burst read timed out
going to wait on addr GAP_RDY
wait on gap_rdy done witg buff ptr 0 469784956
loading image with addr 469784956 addr_min 469784956 and size 262144
load image done

```
The command that fails is the gap8-opoenocd, and specifically, the `script tcl/flash_image.tcl` script.  This is located here:
https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/gap8-openocd-tools/tcl/flash_image.tcl

Line 143 is where it stalls, and this line was changed from 3.8 to 3.9.  Reverting this file does not solve the issue though.

bot-motion, 2021-05-29:
The GAP8 OpenOCD always worked for me 100% in the SDK 3.9.1 docker image. Strange that it fails for you! 

Thanks for clarifying. I started playing around with installing tensorflow as well, installing 'naively' w/o specifying a version, installs tf 1.14. However the SDK relies on TF 1.15 - so errors ensue. First it requires a syntax change to convert the model (some tf API changed) but then it leads to failures some of which I cannot fix. 
```
GENERATING NNTOOL STATE FILE
/gap_sdk/tools/nntool/nntool -s model/nntool_script BUILD_MODEL_SQ8BIT/model.tflite 
```
fails with 
```
File "/usr/local/lib/python3.6/dist-packages/cmd2/argparse_custom.py", line 748, in add_subparsers
    return super().add_subparsers(**kwargs)
  File "/usr/lib/python3.6/argparse.py", line 1716, in add_subparsers
    action = parsers_class(option_strings=[], **kwargs)
TypeError: __init__() got an unexpected keyword argument 'required'
model_rules.mk:64: recipe for target 'BUILD_MODEL_SQ8BIT/model.json' failed
```

The alternative is to force-install tf 1.15 which fails for Ubuntu's standard Python install, b/c the wheels aren't there. So I'm now installing miniconda in the docker container. I'll re-post here if I ever get anywhere. 

I remember that SDK 3.8.1 was a pain in the a* to get running too. Feels like a trip down memory lane.

neoamos, 2021-05-29:
You can install tensorflow 1.15 by doing
```
pip3 install -U pip
pip3 install tensorflow==1.15.2
```

I also had the cmd2 issue you mentioned.  I just edited `/gap_sdk/tools/nntool/interpreter/commands/qtune.py` to remove the `required=True` from line 51.  After that it gets to the flash issue.

neoamos, 2021-08-03:
The flashing issue has been in the current master branch of the gap_sdk (as per [this](https://github.com/GreenWaves-Technologies/gap_sdk/issues/252) issue).  I see that the docker instructions now just point to an official docker image.  It would be good to transition this to the new version, although maybe its better to wait till the next release of the sdk since the issue is only fixed in the master branch currently.

knmcguire, 2021-08-03:
Hi! I guess it's official as it is on docker hub, but it is more out of convenience  and it's good to start separating functionalities from this current repo (as we usually have docker files in their own github repo). But technically the latest docker container on bitcraze/aideck is the exact same file as it was here in the repo :).

You can probably already see here in the [docker repo](https://github.com/bitcraze/docker-aideck) but I already gotten the latest 3.9 and 4.0.0 dockerfile uploaded. We just want to test it more thoroughly before tagging it pushing it officially to docker hub. Once we do that, than it shouldn't be much effort to update it to the latest release that has the fix. Anyway, it should be up soon!

knmcguire, 2021-08-05:
btw, the qtune.py is a difficult one to solve for docker... but I managed to fix it by adding:

`RUN sed -i 's/required=True/ /g' /gap_sdk/tools/nntool/interpreter/commands/qtune.py
`

before making the nntool.

knmcguire, 2021-08-05:
By the way, for big examples, it seems that 'make run' is hanging too. Anybody experiencing that too?

perhaps its not a bad idea to keep on advising to use 3.8.1 ....

knmcguire, 2021-12-15:
This was a bit forgotten but we should try to update the 'latest' docker tags now...

gemenerik, 2024-04-11:
Docker has been updated to latest gap_sdk

closed 2024-04-11 by gemenerik