bitcraze/aideck-gap8-examples #43: AutoTiler link isn't suttibale to the docker image

eladpar-rnd, 2021-04-02:
Hey, the docker image has a Linux Ubuntu 18.04 distribution .
here :
[https://github.com/bitcraze/AIdeck_examples/blob/master/docs/getting-started/docker-gap8.md](url)
it says `This will install the AutoTiler, which requires you to register your email and get a special URL token to download and install the AutoTiler,`

but the email I got says there is only AutoTiler for Ubuntu 16.04

knmcguire, 2021-04-06:
Ah oke, good to know. I wouldn't worry about it and just use the link they provide. I used it last time on the docker and there was no problem with it. Also greenwaves are currently working on 18.04 by standard from their documentation: https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html

eladpar-rnd, 2021-04-26:
i get this error :
`Resolving greenwaves-technologies.com (greenwaves-technologies.com)... 35.197.241.59
Connecting to greenwaves-technologies.com (greenwaves-technologies.com)|35.197.241.59|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-04-26 05:53:56 ERROR 404: Not Found.

Makefile:21: recipe for target 'lib/libtile.a' failed
make[1]: *** [lib/libtile.a] Error 1
make[1]: Leaving directory '/gap_sdk/tools/autotiler_v2'
Makefile:112: recipe for target 'autotiler' failed
make: *** [autotiler] Error 2
`

eladpar-rnd, 2021-05-19:
???

knmcguire, 2021-05-19:
Hi!! Sorry for the late reply! It seems that this is a greenwaves gap SDK
issue. Are you using the gap sdk we advise in the docs: 3.8.1?

By the way, we also have an aideck category on our forum.bitcraze.io in
case u would like a faster answer next time. We check that more often.

eladpar-rnd, 2021-05-23:
we solved the problem
apparently during the installation of autotiler I probably got the link from the site and not the script that uses a post request .
one important lesson is that even if the autotiler link is incorrect it just exits without any error message , I think it is important.
Thanks, Elad.

closed 2021-05-23 by eladpar-rnd

knmcguire, 2021-05-24:
Okay good to know! I submitted an issue to the gap_sdk repo abou this: https://github.com/GreenWaves-Technologies/gap_sdk/issues/227