bitcraze/aideck-gap8-examples #46: On board object detection

semihtasbas, 2021-05-26:
Hi,

Is object detection possible on board on AIdeck? Do you have any test about this? Does it support tf2 model zoo SSD models? Or do you suggest off board detection?

Thank you

knmcguire, 2021-05-27:
Hi! We have a face detector example in our AIdeck examples. And the MNIST example recognizes numbers.

Since the issue is closed, did you find what you were looking for?

semihtasbas, 2021-05-28:
Hi,

Thank you for your interest. After asking the question yesterday, I came across an article, "A 64mW DNN-based Visual Navigation Engine for Autonomous Nano-Drones". In this article, there is an expression about DroNet, "more  than  1 MB  of  memory  needed solely  to  store  the  network’s  weights". When I compared this size with object detection models, I thought it was much smaller. I will try to optimize and deploy a ssd model. But, I don't think, I can achieve it. If you have a knowledge or suggestion, I would be happy to know.

knmcguire, 2021-05-31:
Hi! All I know from now that the PULP platform group are planning to port it to the AIdeck, but this has not been released. You should probably ask again in a few months.

Pafrak, 2021-10-19:
Hello @semihtasbas 
 you can have a look at [NNMenu](https://github.com/GreenWaves-Technologies/nn_menu) repository, here you can find several examples on NN running on GAP8 which some are taken from the TF model zoo, and you have also some examples on how to port an TFlite model onto gap.

knmcguire, 2021-12-15:
Pulpdronet should now be ported to the aideck: https://github.com/pulp-platform/pulp-dronet

I'll close this issue again:) but reopen if the question remains