bitcraze/aideck-gap8-examples #48: Error with GAP8 Docker

rallen10, 2021-05-29:
While trying to [setup the GAP8 Docker container](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/) I run into several errors. 

The first of which I was able to work around, and [created a pull request to fix](https://github.com/bitcraze/AIdeck_examples/pull/47).

The next one I ran into, I haven't been able to sort out yet. After running 

```bash
sudo docker build --tag gapsdk:${GAP_SDK_VERSION} --build-arg GAP_SDK_VERSION=$GAP_SDK_VERSION .
```

I eventually get the error

```bash
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/universe/libk/libkml/libkmlbase1_1.3.0-5_amd64.deb  Undetermined Error [IP: 91.189.88.152 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
The command '/bin/sh -c apt-get install -y libopencv-dev python3-opencv' returned a non-zero code: 100
```

Could this perhaps be fixed with the [new SDK version](https://github.com/bitcraze/AIdeck_examples/issues/41)

knmcguire, 2021-05-31:
Hmm interesting. I haven't seen this before yet and the docker was tried out pretty recently. It seems to be a connection problem. Are you building the docker image from behind a vpn  or in a university network perhaps?

knmcguire, 2021-06-02:
I just build the docker from scratch but I'm not able to reproduce your failed to fetch issue... and I tried it out on the virtual machine, windows WSL 20.04 and a native ubuntu install of 20.04. Could you try again ? maybe there was something wrong with that repo temporarily, perhaps the server was broken.

rallen10, 2021-06-10:
I might have found a fix based on [this stackexchange post](https://askubuntu.com/a/519565/446331) and the [docker file best practices](https://docs.docker.com/develop/develop-images/dockerfile_best-practices/). In short, I added `apt-get update &&` to the line:

```
RUN apt-get update && apt-get install -y libopencv-dev python3-opencv
```

And it seemed to work. I can open a pull request for the change if you like

rallen10, 2021-06-29:
For resolution, please see pull request here: https://github.com/bitcraze/AIdeck_examples/pull/47/files/734dbf46a58ca81df3d0ba643d828fd0f49ac1d3..19442eeb45b05a8095d41680a6e800804b4c2392

closed 2021-06-29 by rallen10