bitcraze/aideck-gap8-examples #54: Record video stream and Crazyflie control inputs

rallen10, 2021-09-17:
I am looking for existing examples of recording the video stream from the AI deck, along side with the control inputs to the Crazyflie. Does something like this exist? I know how to [view the stream](https://github.com/bitcraze/AIdeck_examples/blob/master/NINA/viewer.py), but I want to record it as a data collection

knmcguire, 2021-09-22:
Hi! No examples for this yet... but it shouldn't be difficult I assume? It's using the GTK library, so with some quick googling:
I've found [ this thread](https://stackoverflow.com/questions/39938279/save-current-window-as-image-using-gtk).

Then everytime [the image is updated](https://github.com/bitcraze/AIdeck_examples/blob/a89e29072b7069b819dc930a8ca4a050b026d28d/NINA/viewer.py#L112) the picture can be saved in that manner.

I'll see if I can look at it some time this week but let me know if you managed to get it to work yourself in the mean time.

gemenerik, 2025-06-27:
We have the [FPV example](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/aideck/fpv.py)

closed 2025-06-27 by gemenerik