bitcraze/aideck-gap8-examples #55: Problems running Test Himax Camera

Ramseyous0109, 2021-11-16:
Hello,
I'm studying to use test ai-deck's camera function. When I'm trying to flash the` AIdeck_examples/GAP8/test_functionalities/test_camera` module, my flash program stacks at "Opening Himax Camera". The detailed outputs are as follows.

Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled
Warn : Burst read timed out
	\*\*\* PMSIS Camera with LCD Example \*\*\*
Entering main controller
Testing normal camera capture
Opening Himax camera
^Z

Furthermore, when I flashed and tested the wifi stream example, the `viewer.py` stacked at `socked connected` and no image displayed.

I don't think it's a hardware problem because I've tried on two different ai-deck boards. And I'm wondering if you could give me some idea on how to fix this issue.

Thanks,
David

knmcguire, 2021-11-16:
It seems to freeze when trying to open the Himax camera. Have you tried just restarting the crazyflie (and plugging out it's battery) and just try it again?

Which SDK of the gap8 are you using?

Ramseyous0109, 2021-11-18:
Thanks for your early reply!
I tried restarting the crazyfile and it didn't work.
I'm using the latest version of gap sdk of 4.7.0

knmcguire, 2021-11-18:
The example was last tested for SDK version 3.8: https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/test-functions/wifi-streamer/ 

So it might be due to that. 

So you can downgrade the SDK to that or use [the docker image ](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/)

Ramseyous0109, 2021-11-19:
I used the docker and it worked!
Thank you for your help!

ChenMathilda, 2021-11-19:
> I used the docker and it worked! Thank you for your help!

hi,
I met the same error like you! I want know do you solved it?
All of  my 3 team member can apply this experiment, but I always failed. Could this problem be attributed to the computer? I have a teammate who couldn't do it at the beginning. After restarting the computer and executing the program, it succeeded.
By the way,  my GAP SDK is 3.8.1.

ChenMathilda, 2021-11-19:
> Thanks for your early reply! I tried restarting the crazyfile and it didn't work. I'm using the latest version of gap sdk of 4.7.0

Hi,
I met the same error like you! I want know do you solved it?
All of my 3 team member can apply this experiment, but I always failed. Could this problem be attributed to the computer? I have a teammate who couldn't do it at the beginning. After restarting the computer and executing the program, it succeeded.
By the way, my GAP SDK is 3.8.1.

Ramseyous0109, 2021-11-19:
> > Thanks for your early reply! I tried restarting the crazyfile and it didn't work. I'm using the latest version of gap sdk of 4.7.0
> 
> Hi, I met the same error like you! I want know do you solved it? All of my 3 team members can apply this experiment, but I always failed. Could this problem be attributed to the computer? I have a teammate who couldn't do it at the beginning. After restarting the computer and executing the program, it succeeded. By the way, my GAP SDK is 3.8.1 and I used[ the docker images ](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/)

Hello, I just switched to the [ the official docker images ](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/) and it successfully worked. So perhaps it's not the same reason for you. Maybe the official staff can help you out.

ChenMathilda, 2021-11-19:
> Hello, I just switched to the [ the official docker images ](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/) and it successfully worked. So [...]

Hi,
 The official docker images envirment is exactly what I used.
And I also solved this problem just know: I find I didn't make sdk & make gap_tools in docker.
Thanks for your reply!
Regards.
Mathilda

knmcguire, 2021-11-19:
Ah good you both were able to fix this! The make sdk and make gap_tools requirement should have not mattered for the wifi camera example or the test camera example... but if it works it works right! 

I'll close this issue:)

closed 2021-11-19 by knmcguire