bitcraze/aideck-gap8-examples #56: Testing the Himax camera

ChenMathilda, 2021-11-24:
Hi,
When I try to run the "Testing the Himax camera", it will always stop at certain point like below:

[image]

Therefore, I try to find the program stop at which scentence,and find:

 [image]

But 3 of my teammates can run this example successfully, and we have the same virtual envirnment.
 So, I want to know is this problem may be caused by computer configuration?  or there are other causes?
By the way, my teammates and I all used the offical docker. And we refer to this page:https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/test-functions/test-camera/
Regards,
Mathilda

knmcguire, 2021-11-24:
Ah yeah let's see what's going on!
As you illuminated your computer setup being the issue, there might be some other things going on:
- Your crazyflie does not have the aideck driver initialized according to these instructions: https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/tryout/
- The camera module might not be connecting properly or is broken. Maybe it would be good to try out the camera modules of the other AIdecks perhaps ?  https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/guides/camera-replace/

knmcguire, 2021-11-26:
Just an update, this issue seems to be resolved on the forum: https://forum.bitcraze.io/viewtopic.php?t=5166
Glad you got it to work! @ChenMathilda 

Just for future reference, make sure to notify in the threads on both forum and github if there is any double posts of your question, just to avoid some confusion for us. As long as you say that you also asked the same question on either platform that can already be helpful

closed 2021-11-26 by knmcguire