bitcraze/aideck-gap8-examples #57: Can NN example use picture captured from camera real time, how can I do it?

aqqz, 2021-12-14:
I tested the himax camera demo and image classification example, can this two demos use together? Now picture is set by Makefile, if I want to capture pictures and inference it onboard, shall I use L3 MEM on the AIdeck? I search pulp_dronet codes but not find camera loop, thanks for your reply.

knmcguire, 2021-12-14:
Yes that should be possible, but you will need to make the app program yourself. You can look at the face detection example for instance, which does not use an NN but it does show an example of the cluster and the video streamer simultaneously, if you enable the right defines in the Makefile (see t[his guide ](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/img-proc-examples/face-detection/). 

About the pulp-dronet, you should probably make an issue on their repository with this question if you haven't already.

closed 2021-12-15 by knmcguire

aqqz, 2021-12-16:
OK, I write a demo program implements it. Another question, the color of RGB888_IO output image is approaching green, how to solve it ?