bitcraze/aideck-gap8-examples #58: Native Installation on Ubuntu 20.04

whoenig, 2021-12-15:
I tried following the instructions at https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/getting-started/ on Ubuntu 20.04. I tried different versions of the SDK (3.8.1, 4.0.0) and usually get a 

```
cc1plus: error: bad value (‘tigerlake’) for ‘-march=’ switch
```
when making the SDK.

For the steps, I oriented myself at the docker instructions in https://github.com/bitcraze/docker-aideck/blob/master/src/Dockerfile as well as the readme for the official SDK.

Is the recommended way here to use Docker, or did anybody succeed at using a native installation? The current Docker image uses SDK 4.0.0 and not 3.8.1 as stated in the documentation, correct?

knmcguire, 2021-12-15:
hi! I believe the 'latest' docker image is still on 3.8.1, as that downloads by standard, since some examples still had some problems with the real latest version of docker. You can check this by 
`docker run -it bitcraze/aideck`
and 
```
cd ../../gap_sdk
git status
```
if you want the latest version that is available on docker, you can try the :4.0.0 after the container name.

whoenig, 2021-12-15:
Thanks, Kimberly!

closed 2021-12-15 by whoenig