bitcraze/aideck-gap8-examples #60: Manual exposure control

zeroos, 2022-01-10:
Hey guys!

I was trying to work with AI-deck and the himax camera without the AEG turned on. We'd like to be able to fly the crazyflie in very low lighting conditions and wanted to turn off the AEG, fix integration time (exposure), analog gain, and digital gain to have more reproducible images.

However, it seems like changing the appropriate parameters does not result in any change in the captured images. I've created an MWE to better visualize what I mean:
https://github.com/bitcraze/AIdeck_examples/compare/master...zeroos:mwe/manual_exposure?expand=1
I would expect that running this code would result in images with constantly changed exposure, but all the images are exactly the same.

Am I missing something or is it a bug in HIMAX?

knmcguire, 2022-01-12:
Hi!

The driver for the camera is actually part of the [gap_sdk](https://github.com/GreenWaves-Technologies/gap_sdk). Have you looked at their issue list if this is a problem?

What I do know that is is important to set the registries before the PI_CAMERA_CMD_STOP as that puts the camera is sleep mode and for some reason makes it irresponsive to registry changes... It seems to be a known bug in the driver. You can perhaps look at the [wifi example in these lines](https://github.com/bitcraze/AIdeck_examples/blob/9bb9b1c157645e877db9677bc35f383c6dc667dd/GAP8/test_functionalities/wifi_jpeg_streamer/test.c#L71-L82) for a reference for that. Perhaps that is what is going on?

knmcguire, 2022-01-12:
this change was related to #51 btw

zeroos, 2022-01-14:
Thanks for the suggestion! In fact, I am changing the parameters between PI_CAMERA_CMD_START and PI_CAMERA_CMD_STOP commands... In the same way, I was changing some parameters related to automatic exposure control (with AEG ON) and it was working fine, so I thought these parameters should also be changeable. 

In fact, what I'm trying to do right now (as a workaround) is to leave AEG on and change `HIMAX_MIN_DGAIN` and `HIMAX_MAX_DGAIN` to the same value I wanted to set `DGAIN` to and do the same for other parameters. It sort of works, but I'm still experimenting trying to find out if that approach is reliable.

Anyways, thanks for the suggestion! I've created an issue with gap_sdk [https://github.com/GreenWaves-Technologies/gap_sdk/issues/301], so I think this one can be closed.

closed 2022-01-14 by zeroos