bitcraze/aideck-gap8-examples #63: Automatic exposure gain control doesn't work

neoamos, 2022-02-24:
From what I understand, the Himax camera should be able to adjust the exposure continuously to adapt to different lighting conditions.  In the WiFi streaming example, this is not the case.  If you turn on the camera in a dark room, and then turn the lights on, the image will be completely washed out and the camera does not adjust the new lighting conditions.  If you start the camera in a room that is already bright, the exposure will be set correctly.  It seems that the exposure is only set at the time the camera is turned on, and then it stays the same.
I have the AI deck v1.1

neoamos, 2022-03-01:
I found a workaround to this issue is to read a 324x324 image before reading the actual image you want.  This seems to force the camera to update the exposure.  It does lead to a bit of flickering, but it is worth it for to get dynamic exposure.

https://github.com/neoamos/AIdeck_examples/commit/e77e0424413d68c2469e11639740470a05cdbe11

williamleong, 2024-01-02:
The solution by @neoamos doesn't appear to work for me but I am not sure if it is due to some change since it was last posted or maybe it was something I did. Maybe someone else has a better way to do this now?

whoenig, 2024-01-02:
Not sure if it helps, but we had a hard time figuring out how to disable automatic exposure. Our solution is at https://github.com/IMRCLab/aideck-gap8-examples/blob/1da6336c290da14c9be7f9b298835c09084acbbe/examples/other/wifi-img-streamer/wifi-img-streamer.c#L105-L121. Perhaps first disabling and then re-enabling every second or so will work?

williamleong, 2024-01-03:
Thanks @whoenig. I've tried that with the face detection example, using `pi_camera_control(&cam, PI_CAMERA_CMD_AEG_INIT, 0);` after every capture, but something else is turning the automatic exposure off. If I try to view the stream, what I'd see is that the image brightens for a moment with the correct exposure, then turns dark again as if automatic exposure is turned off.

Will try a few more things and report back if I find anything.

williamleong, 2024-01-03:
After some testing, this seems to work for setting the auto exposure in the face detection example:

```
// Use the following for automatic flicker control
set_register(0x2100, 0x1);  // AE_CTRL
// This is needed for the camera to actually update its registers.
set_register(0x0104, 0x1);

pi_camera_control(&cam, PI_CAMERA_CMD_START, 0);

pi_camera_capture(&cam, imgBuff0, CAM_WIDTH*CAM_HEIGHT);

// Use the following for automatic flicker control
set_register(0x2100, 0x1);  // AE_CTRL
// This is needed for the camera to actually update its registers.
set_register(0x0104, 0x1);

pi_camera_control(&cam, PI_CAMERA_CMD_STOP, 0);

// Send task to the cluster and print response
pi_cluster_send_task_to_cl(&cluster_dev, task);
```

Not sure why it needs to be set twice but I tried various combinations and it looks like this is working for me.

Another unrelated observation is that the `pi_camera_capture` seems much more stable than the original method with `pi_camera_capture_async`, I am not sure why, but I guess that's a problem for another issue.

knmcguire, 2024-01-03:
Hmmm that is indeed odd that it needs to be set twice. I would imagine that perhaps if you do it too early that the registries won't go through well... So if you remove the first registry-set it doesn't work for the second one?

knmcguire, 2024-01-10:
Hmmm that is perhaps a better question for the Greenwaves folks I'm afraid... I'm not sure what the function does either.

williamleong, 2024-08-05:
Just some update - I was still having trouble and it seems the most reliable way is to set the exposure after `pi_camera_control(&camera, PI_CAMERA_CMD_START, 0);`:

```
pi_camera_control(&camera, PI_CAMERA_CMD_START, 0);
//enable auto exposure here
pi_camera_capture(&camera, imgBuff, CAM_WIDTH*CAM_HEIGHT);
pi_camera_control(&camera, PI_CAMERA_CMD_STOP, 0);
```
The image will oscillate between a darker and brighter image if the brightness is on some threshold, I'm guessing that the exposure gain control is very coarse (ANALOG_GLOBAL_GAIN setting only has options like 2x, 4x).

gemenerik, 2026-10-19:
This appears resolved. Additionally, auto-exposure is available and configurable for NanoCockpit. For example, check out the  [NanoCockpit streamer](https://github.com/bitcraze/aideck-gap8-examples/blob/master/docs/examples/nanocockpit/streamer.md).

closed 2026-10-19 by gemenerik