bitcraze/aideck-gap8-examples #70: struct.error while using opencv-viewer.py

matze946, 2022-04-07:
Hey :)
I tried to use the opencv-viewer.py  and got a struct.error message 

`Connecting to socket on 192.168.4.1:5000...
Socket connected
Traceback (most recent call last):
  File "streamer.py", line 86, in <module>
    [magic, width, height, depth, format, size] = struct.unpack('<BHHBBI', imgHeader)
struct.error: unpack requires a buffer of 11 bytes`

Any ideas how to solve this?

knmcguire, 2022-04-07:
Hi! Did you update your wifi example with the latest example? or are you trying it out right out of the box? You would probably need to revert the examples to this commit: bdcca78839a0e61baee97e610d9c73b91d9934e8 in that case.

matze946, 2022-04-08:
Yes I first used it without updating to the latest example. But after updating the examples the error is still the same..

evoggy, 2022-04-12:
I haven't seen this before. Do you get this the first time you connect as well? Or is it only after disconnecting/connecting again?

kjayarajah, 2022-05-06:
Anybody have any luck with this? I'm having this issue even after going back to the older commit. Thanks in advance for any help -- been stuck with this for several days.

matze946, 2022-05-06:
I updated everything again like in the getting started pages [https://www.bitcraze.io/documentation/tutorials/getting-started-with-aideck/]
That solved the problem for me, but now I am still not able to stream the images with the opencv-viewer.py 
It says that I am connected to the socket, but I can't see the viewer.
Any advice on that issue?

knmcguire, 2022-05-06:
Good you fixed that build problem.

Were you able to flash over air the wifi example okay without error? Are you connected to the aodeck as accesspoint or is it connecting to wifi itself?

matze946, 2022-05-06:
Yes I was able to flash the wifi example using the radio without any errors and I'm connected with the aideck as accesspoint.

knmcguire, 2022-05-06:
Ah oke. Sometimes I had to powercycle my crazyflie a time or two before it was connected. You can also have feedback from the console tab of the Cfclient for any cpx readouts. Usually it would say if it manage to establish a connection or if anything failed. If the camera failed to initialized for instance, you'll see that in the console tab.

knmcguire, 2022-05-09:
ah great! we are trying to work on the stability of it all, as it should not be necessary to constantly restart it... but I'll keep it open here so that other people know of it if they are seeing the same thing of what they should do.

gemenerik, 2026-10-19:
We have not received reports since. Closing the issue. It can still be found after closing.

closed 2026-10-19 by gemenerik