bitcraze/aideck-gap8-examples #83: Is there an exmple of how to use the streamer at the same time as the classification demo? Has it been done?

moaarv, 2022-04-21:


moaarv, 2022-04-21:
We are trying to make the streamer work at the same time as we use the classification demo on our crazyflie 2.1. Is there any way of doing this?

knmcguire, 2022-04-26:
The facedetection example uses the streamer: https://github.com/bitcraze/aideck-gap8-examples/tree/master/examples/image_processing/FaceDetection 

Do mind that we are right in the process of converting the examples to the new CPX protocol... so maybe it would be worth to wait a few weeks until this has been fixed.

gemenerik, 2022-05-02:
Can't wait to use it! I see that the CPX protocol is based on FreeRTOS, are you considering PUPLOS support? Or I suppose that discussion is part of converting the examples?

moaarv, 2022-05-02:
Do you know how often the classification demo classifies per second?

knmcguire, 2022-05-02:
I think it is about 1 or 2 hz for the classification example. It is based on mobile net v2 which is the biggest network pretty much the AIdeck can handle

@gemenerik good question! No plans yet I think, but perhaps better to start a separate thread about this for now as it is not really related to the original post :) It would be good to continue the discussion!

gemenerik, 2022-05-02:
I was curious because I think the CPX and the streamer being based on FreeRTOS and the classification demo being based on PULPOS is the biggest problem you will run into when trying to merge them, otherwise it should be quite trivial.

It's probably good to make a separate issue of it yes!