bitcraze/aideck-gap8-examples #84: img-viewer example or docs broken?

whoenig, 2022-04-21:
I tried to update to the latest version of the wifi-img streamer example and didn't get too far, unfortunately.

1. The docs for upgrading the bootloader seem to be outdated in https://www.bitcraze.io/documentation/tutorials/getting-started-with-aideck/. When following the instructions in https://github.com/bitcraze/aideck-gap8-bootloader it seems to work.
2. The instructions in https://www.bitcraze.io/documentation/repository/aideck-gap8-examples/master/test-functions/wifi-streamer/ are
```
$ cd aideck-gap8-examples
$ docker run --rm -v ${PWD}:/module bitcraze/aideck tools/build/make-example examples/other/test_functionalities/wifi_jpeg_streamer image
```
This results in an error:
```
docker: Error response from daemon: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "tools/build/make-example": stat tools/build/make-example: no such file or directory: unknown.
```
In CI this seems to work, so I am not sure what the issue might be (Ubuntu 20.04).

evoggy, 2022-04-21:
Thanks for the feedback, will look into this.

evoggy, 2022-04-21:
1) I tried re-running the building/flashing of the GAP8 bootloader and it seems to be working. Did you mean these instructions: https://www.bitcraze.io/documentation/tutorials/getting-started-with-aideck/#flash-the-latest-gap8-bootloader ?

2) Unfortunately the path in the repository is out of sync on the web, we moved the example earlier today. We'll deploy a new version that will reflect the way it looks now.

evoggy, 2022-04-21:
2) Thanks for noticing this, a new version has been deployed now: https://www.bitcraze.io/documentation/repository/aideck-gap8-examples/master/test-functions/wifi-streamer/

whoenig, 2022-04-21:
1. Yes. The main difference between the instructions (web vs. github repo) is `bitcraze/aideck:4.8.0.1` vs `bitcraze/aideck:latest`. For me it was important to use the explicit `4.8.0.1` version. Not sure to which version `latest` exactly points to.
2. The slightly different path wasn't a big issue and easy to fix. The problem I am having is that docker can't find `tools/build/make-example` (even though it is there in the local system). Since it works fine in CI, I am wondering if there is some magic docker trick I am missing (clean up, or sync the docker containers etc.)?

evoggy, 2022-04-21:
Yeah, the answer to both issues might be that the CI always pulls the latest docker, i.e doing ```docker pull bitcraze/aideck```. The latest version is 4.8.0.2, although the changes from 4.8.0.1 shouldn't effect this. But the changes from 4.8.0 to 4.8.0.1 will.

I've recently learned that using ```bitcraze/aideck:latest``` only uses the latest local one and doesn't actually pull anything new.

If you do a ```docker pull bitcraze/aideck``` and try the building for the WiFi example does this work?

evoggy, 2022-04-21:
After doing some testing here (and me learning about docker) the issue seems to be that the ```latest``` tag doesn't necessarily point at the latest version... So what's probably happening is that you had 4.8.0 before (and this has the tag ```latest```), then copy--pasting the line from the turorial explicitly pulls 4.8.0.1 and uses this. But at this point the test ```latest``` is still pointing at 4.8.0, so when you copy past the next line this will use ```bincraze/aideck:latest``` which is 4.8.0 and this does not work with building the examples. Thanks for noticing this!

I think you will solve your problem by doing ```docker pull bitcraze/aideck``` which will update the ```latest``` tag and we will update the documentation to explicitly say that you should do the pull before running the examples.

whoenig, 2022-04-21:
That did the trick! Thanks so much, Marcus. Perhaps we can add this to the CI as well, to make it clear what a user has to do.

closed 2022-04-21 by whoenig