bitcraze/aideck-gap8-examples #91: Issues with the classification demo

knmcguire, 2022-05-09:
Soon PR #89 will be merged but there are still issues with it though.

- CPX message queue needed to be reduced from 80 to 5, or else the example stalls
- Setting the fabric controller or cluster frequency higher made the example stall
- CPX can only print one message. 

This all still needs to be investigated

aqqz, 2022-05-25:
Is the current ai example must run on the sdk4.7 version, I use the lastest 4.12 version can not run it, it seems memory problems.

knmcguire, 2022-05-25:
It's based on the bitcraze/aideck docker container, which is SDK 4.8. Interesting that there are issues with the latest SDK! That really means that this example is right on the border of what it can handle together with the bootloader and CPX. 

Did you flash the example over air or with a programmer (so without the bootloader) ?

aqqz, 2022-05-25:
Yes, I have not test the bootloader ways, test it on the jtag progarmmer.

eladpar-rnd, 2022-08-14:
I couldn't run the example on SDK 4.12 (as mentioned in closed issue)
but reverting to GAP SDK 4.8.0 solved the problem

in the mean time, i have finished compiling the example (got the img file)
but the flashing part dosnn't work:
[image]

Is it possible to get a few words on how to flash the GAP8 with the .img file via the JTAG?
@aqqz

knmcguire, 2022-08-15:
@eladpar-rnd Your followup questions is not related to this issue. Could you please post the question in Q&A of https://discussions.bitcraze.io/

knmcguire, 2022-11-30:
Just a comment, there has been a thought about cpx not being thread safe, so that it is actually not about the size of the networks. See this ticket: https://github.com/bitcraze/aideck-gap8-examples/issues/100

closed 2023-02-06 by hmllr