bitcraze/aideck-gap8-examples #96: make menuconfig error

vanbong, 2022-05-20:
bong@bong-IRH-Gaming:~/crazyflie-firmware$ make menuconfig
make[1]: Entering directory '/home/bong/crazyflie-firmware/build'
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/mconf.o
<command-line>: fatal error: curses.h: No such file or directory
compilation terminated.
make[2]: *** [scripts/Makefile.host:108: scripts/kconfig/mconf.o] Error 1
make[1]: *** [tools/kbuild/Makefile.kbuild:403: menuconfig] Error 2
make[1]: Leaving directory '/home/bong/crazyflie-firmware/build'
make: *** [tools/kbuild/Makefile.kbuild:150: sub-make] Error 2

what is problem for me?

knmcguire, 2022-05-23:
It seems that you need to install the curses library. The [wifi example repo ](https://www.bitcraze.io/documentation/repository/aideck-gap8-examples/master/test-functions/wifi-streamer/)doc said:

_First see the [crazyflie-firmware repository documentation](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/building-and-flashing/build/) for more details on how to build and flash and checkout how to setup [kbuild’s menuconfig](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/kbuild/)._

and the [kbuild menuconfig](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/development/kbuild/) noted that should install:

```
sudo apt install build-essential libncurses5-dev
```

but perhaps this should have been noted more clearly... We will rewrite this a bit!

gemenerik, 2026-10-19:
Stale, incorrect repository