bitcraze/aideck-gap8-examples #97: Flash Wifi hanging

simutisernestas, 2022-05-29:
In the getting started with AI Deck guide [here](https://www.bitcraze.io/documentation/tutorials/getting-started-with-aideck/) the command:
```cfloader flash aideck_gap8_wifi_img_streamer_with_ap.bin deck-bcAI:gap8-fw -w [CRAZYFLIE_URI]```
Hangs infinitely at 2%. Do you have any idea what could cause the issue?

The output:
```
$ cfloader flash aideck_gap8_wifi_img_streamer_with_ap.bin deck-bcAI:gap8-fw -w $CRAZYURI
Reset to bootloader mode ...
Could not save cache, no writable directory
Could not save cache, no writable directory
Skipping bcAI:esp, not in the target list
Deck bcAI:gap8, reset to bootloader
| 0% Writing to bcAI:gap8 deck memory
/ 1% Writing to bcAI:gap8 deck memory
- 2% Writing to bcAI:gap8 deck memory
```

And after an hour interrupting with keyboard:
```
^CTraceback (most recent call last):
  File "/home/ernis/.local/bin/cfloader", line 8, in <module>
    sys.exit(main())
  File "/home/ernis/.local/lib/python3.8/site-packages/cfloader/__init__.py", line 155, in main
    bl.flash_full(None, filename, warm_boot, targets)
  File "/home/ernis/crazy/crazyflie-lib-python/cflib/bootloader/__init__.py", line 221, in flash_full
    self.flash(filename, targets, cf)
  File "/home/ernis/crazy/crazyflie-lib-python/cflib/bootloader/__init__.py", line 173, in flash
    current_index = self._flash_deck_incrementally(deck_artifacts, deck_targets, current_index)
  File "/home/ernis/crazy/crazyflie-lib-python/cflib/bootloader/__init__.py", line 493, in _flash_deck_incrementally
    result = deck.write_sync(0, deck_artifact.content, progress_cb)
  File "/home/ernis/crazy/crazyflie-lib-python/cflib/crazyflie/mem/deck_memory.py", line 85, in write_sync
    syncer.wait()
  File "/home/ernis/crazy/crazyflie-lib-python/cflib/utils/callbacks.py", line 77, in wait
    self._event.wait()
  File "/usr/lib/python3.8/threading.py", line 558, in wait
    signaled = self._cond.wait(timeout)
  File "/usr/lib/python3.8/threading.py", line 302, in wait
    waiter.acquire()
KeyboardInterrupt
```

simutisernestas, 2022-05-29:
No worries, solved it. The issue was that i was passing address with the URI. If anyone is struggling with it replace `radio://0/80/2M/E7E7E7E7E7` with `radio://0/80/2M` : )

closed 2022-05-29 by simutisernestas

knmcguire, 2022-05-30:
Hi! I'm converting this to our [discussions](https://discussions.bitcraze.io/) as this is not only AIdeck related, it is an fault in our cfloader which is part of the cfclient repo.