bitcraze/aideck-gap8-examples #104: [AI-Deck] Any insights on cfloader flash img hangs at 99% specifically?

knmcguire, 2022-11-30:


knmcguire, 2022-11-30:
So after investigations it seems that any binary file larger than 747776 Bytes fails with the cfloader (getting stuck at 99%), and the classification example is 747792 Bytes large!

This was possible to flash before... so what changed? The docker file itself hasn't changed, but perhaps the autotiler that needs to be pulled with that version is updated somehow?

chengliu-LR, 2022-12-01:
I also tried this binary file (can be downloaded through the following Google Drive link):
https://drive.google.com/file/d/1n7XXOmcSYcQdgXulDZKMPk2j9ynTmhyN/view?usp=sharing

This binary file is a CNN pytorch neural network, quantized by nemo and compiled by [dory](https://github.com/pulp-platform/dory). It seems that although it is `993KB` large (exceeds 747776 Bytes), it can still be flashed by cfloader. So the file size might not be the issue.

knmcguire, 2022-12-01:
Hmm interesting! Where are you currently with the firmware of the aideck+ crazyflie? release 2022.09 or latest development branch? Will try this binary out soon. 

Yes I agree, it would be really weird if the filesize was the case, as it should be able to handle 60 MB or something like that?

knmcguire, 2022-12-01:
Oh wait I see that you answered my question on https://github.com/orgs/bitcraze/discussions/369 already. 
So your firmware is on commit 5c4261e5a9ff (CLEAN) but you've tried 2022.5 and 2022.9 as well.

hmmmm really strange! The plot thickens... but we need to investigate this more. Thanks for sharing though!

krichardsson, 2022-12-02:
Other examples are

1000 bytes works
2000 bytes fails

closed 2022-12-05 by krichardsson