bitcraze/aideck-gap8-examples #105: Flashing of GAP8 sometimes stops at 3-4%

krichardsson, 2022-12-07:
Sometimes the flashing process is stopping around 3-4% when flashing a binary to the GAP8 from the cfloader. 
I have not investigated this deeply, but it seems as if the GAP8 is blocking for some reason. From a STM fw point of view, buffers are pushed to CPX, and the expected behavior is that the ESP passes them on to the GAP8 where they should be written to flash. The first incoming buffer will also trigger an erase of the first flash page. 

3-4% is most likely the amount of data that fits into the CPX buffers between the STM and GAP8, that is the GAP8 is not consuming (writing data to flash) as expected. When the CPX buffers are full, the STM can not write any more which blocks the memory mapping task. When the memory mapping task stops sending back ACK packets to the python lib, the memory mapping write in the python lib stops. This is finally what stops the cfloader.

We recently added asserts in the STM fw that are triggered [if a write to the GAP8 takes too long](https://github.com/bitcraze/crazyflie-firmware/pull/1160), one of them should reboot the Crazyflie when this happens.
It can be a bit tricky to actually get the assert information out in this scenario, but there are two options
1. Connect to the CF with the python client and click the "Assert Info" button in the console log
2. Set `enable_console_lo` to `True` [in the deck flashing code](https://github.com/bitcraze/crazyflie-lib-python/blob/def4331077947993ffd860dc40c7e389e4d64aef/cflib/bootloader/__init__.py#L401-L402)

whoenig, 2022-12-07:
I have one AI-deck where this is very consistently the issue. This AI deck has the latest ESP firmware (flashed via cfclient update) and gap8-bootloader (flashed via JTAG). When flashing, cfloader hangs for a while and the Crazyflie reboots. The assert information (via cfclient Assert info) is surprisingly an unrelated assert (note that the firmware does not crash using the 2022.09 release):

```
SYS: Assert failed at ../src/hal/src/radiolink.c:164
```

krichardsson, 2022-12-07:
We just got this problem in the stability lab, but the assert was at line 136 in aideck.c

krichardsson, 2023-11-08:
@whoenig do you still see this problem?

whoenig, 2023-11-08:
No - we have only used the AI deck with the JTAG adapter lately.

whoenig, 2024-02-06:
Had this again today with an assert in `stream_buffer.c` (unfortunately didn't write down the exact line). There was no way out, except of flashing the GAP8 directly via JTAG.

knmcguire, 2024-02-07:
hmm this is strange... We need to find time at one point to stabilize the aideck again but it is not really on our priority list right now. We now say for everybody that a jtag programmer is crucial.

gemenerik, 2024-02-19:
gap8 flashing stops at ~6% when having a flow deck attached

williamleong, 2024-03-20:
I find that the success rate seems to be higher if we remove the multiranger and flow deck. Maybe that stops the code in the app layer from running and frees up the CPU.

gemenerik, 2024-05-30:
https://github.com/bitcraze/crazyflie-firmware/pull/1381 fixes OTA flashing freezing for aideck+flowdeck+multiranger

gemenerik, 2024-07-09:
Still happens for aideck+lighthouse. Assert `aideck.c` ln 136

gemenerik, 2024-07-10:
The lighthouse deck seems to add capacitance on VCOM, causing the GAP8 to not reset correctly

gemenerik, 2024-07-10:
As a workaround, we tried a 4.5 kOhm resistor between VCOM and ground. This configuration drops the voltage quickly enough to successfully flash with the lighthouse attached. Note that we are using long wires, which might influence the outcome. Users may experiment with resistances around this range to achieve similar results.

gemenerik, 2024-07-10:
One neat way of doing this would be to put the resistor between the relevant pads on the battery holder.

closed 2025-06-27 by gemenerik