bitcraze/aideck-gap8-examples #116: cpx array sending example to crazyflie

knmcguire, 2023-02-27:
There have been several request to add an example of CPX sending an array to the crazyflie app, and this is one of them.
- https://github.com/orgs/bitcraze/discussions/572

So it would be great to have that in the repository. 

In the next few weeks we will not have time for this (we as in Bitcraze), but if anybody has an example working, perhaps could make a PR?

matthewoots, 2023-03-01:
@knmcguire I do have an example but it is abit messy since it is compiled to the `wifi streamer example` since I don't really know how to create another example and compile with the aideck using docker.
Maybe you can let me know the steps and I can help to do a pull request here.
[image]

hmllr, 2023-03-02:
I just added a tutorial about how to write your own applications on gap8, as this might be an issue for more than just you: https://github.com/bitcraze/bitcraze-website/pull/155 
The example code for it is linked in the PR linked above.
@matthewoots it would be cool if you would use this starting point to do a PR for your CPX communication in both directions :)

matthewoots, 2023-03-02:
Thank you so much for developing the guide @hmllr, I'll get to the example as soon as I can.

Also there are some things that may need to be ironed out in `crazyflie-firmware` the receiving end, since there may be a need to add a module added to `cpx.h` (or for me I tapped on `aideck`) to write the handler function
- Currently it is using `cpx.c` where `cpxInternalRouterReceiveOthers` will receive data with a `CPX_F_TEST` function, in which one can add a data handler inside the case

Moving forward I think I can create a module and attach it to `cpx.h`, this will handle the message and pass it into an empty task (which the user can define?)

hmllr, 2023-03-03:
Hmm, I am not completely sure but I thought that using the CPX_F_APP should send it to the app directly. 
However, as you have something working maybe you can draft a PR and we go from there?

pladosz, 2023-03-29:
Hi,
I have example of how to send CRTP velocity commands from AI-deck over CPX to the crazyflie. It is based on modified gap8-examples code and disucssion from here: [https://github.com/orgs/bitcraze/discussions/519](url). It can be found here: [https://github.com/pladosz/AI_deck_vel_send_cpx](url)

closed 2026-10-19 by gemenerik