bitcraze/aideck-gap8-examples #117: Unable to resolve host address to greenwaves?

matthewoots, 2023-03-01:
In the docker, I don't know whether anyone is facing this issue but I cannot connect to `greenwaves`, it seems to be down?
```
Do you agree with the license (y/n) ? y
--2023-03-01 02:19:06--  https://greenwaves-technologies.com/autotiler/libtile.4.2.0.a
Resolving greenwaves-technologies.com (greenwaves-technologies.com)... 
failed: Temporary failure in name resolution.
wget: unable to resolve host address 'greenwaves-technologies.com'
make[2]: *** [Makefile:23: Autotiler/LibTile.a] Error 1
make[2]: Leaving directory '/gap_sdk/tools/autotiler_v3'
make[1]: *** [gap8/gap8.mk:258: autotiler] Error 2
make[1]: Leaving directory '/gap_sdk'
make: *** [Makefile:42: all] Error 2
```
However, if I do `wget` on my comp it does not should any error
```
wget https://greenwaves-technologies.com/autotiler/libtile.4.2.0.a
--2023-03-01 10:32:30--  https://greenwaves-technologies.com/autotiler/libtile.4.2.0.a
Resolving greenwaves-technologies.com (greenwaves-technologies.com)... 35.197.241.59
Connecting to greenwaves-technologies.com (greenwaves-technologies.com)|35.197.241.59|:443... connected.
HTTP request sent, awaiting response... 307 Temporary Redirect
Location: https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a [following]
--2023-03-01 10:32:32--  https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a
Resolving gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)... 52.219.72.188
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|52.219.72.188|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 976402 (954K) [application/x-archive]
Saving to: ‘libtile.4.2.0.a’

libtile.4.2.0.a     100%[===================>] 953.52K   973KB/s    in 1.0s    

2023-03-01 10:32:34 (973 KB/s) - ‘libtile.4.2.0.a’ saved [976402/976402]

```

hmllr, 2023-03-02:
Hi :)
This issue seems not related to the aideck directly, but to greenwaves. Maybe @Yaooooo can help you with this, or you can ask the question in the gap-sdk issues (https://github.com/GreenWaves-Technologies/gap_sdk/issues).

Yaooooo, 2023-03-02:
Thanks for reporting this, but as you said, when directly copy this path to browser, it works... so it seems the server was down just when you 1st time download it and it recoveried after. 
Could you just do a **make autotiler** to re-trigger the download? If the file libtile.4.2.0.a is already there, please remove it before.

knmcguire, 2023-03-13:
@matthewoots could you confirm your issue being resolved? if so, please close this issue

krichardsson, 2023-03-22:
Please reopen this issue if the problem is not resolved

closed 2023-03-22 by krichardsson