bitcraze/aideck-gap8-examples #132: Himax camera  does not work at lower resolutions

fed12345, 2023-07-18:
Wifi stream goes black if the camera format is set to `cam_conf.format = PI_CAMERA_QQVGA;`, and the width and height of the camera are set to 164 and 124 respectively. On the [website](https://store.bitcraze.io/products/copy-of-ai-deck-color-camera-module) it mentions the camera should support QQVGA.

knmcguire, 2023-07-26:
Hi! Also a good way is to ask this question for the GAP sdk folks as that is where the support is coming from. This has worked before so perhaps there is a bug

https://github.com/GreenWaves-Technologies/gap_sdk

gemenerik, 2026-10-19:
I recommend trying the NanoCockpit examples for which I can confirm QQVGA mode works. For example, see the [NanoCockpit streamer](https://github.com/bitcraze/aideck-gap8-examples/blob/master/docs/examples/nanocockpit/streamer.md). It is configured for QQVGA by default, but you can change format.

closed 2026-10-19 by gemenerik