bitcraze/aideck-gap8-examples #142: Turning off/on the camera

xiaoshan-lin, 2024-02-26:
Hello,

We use the opencv-viewer.py to get the image from aideck. Wondering if there is any way we can send commands to the aideck to turn off and turn on the camera? Thanks in advance for your help and support.

gemenerik, 2024-02-27:
Do you intend to send these commands from the pc?

In that case you can use the [app channel](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/userguides/app_layer/#app-channel-packet-based-communication-between-the-crazyflie-and-the-python-lib) to send data (in this case a camera command) to a Crazyflie app. For the communication from the app to the AI-deck check out the [STM-GAP8 CPX communication example](https://www.bitcraze.io/documentation/repository/aideck-gap8-examples/master/other-examples/stm_gap8_cpx/). Based on the incoming data you can use `pi_camera_control` on the GAP8 side to start/stop the camera.

xiaoshan-lin, 2024-03-06:
Yes, we want to send the commands from the PC. We found a Python example in [examples/app_appchannel_test/ ](https://github.com/bitcraze/crazyflie-firmware/tree/master/examples/app_appchannel_test).

Is the appchannel supported in the cpp Crazyflie Class, or it is only available in Python cflib?

knmcguire, 2024-03-06:
Hi! The link that you shared is actually the [c-based app implementation for onboard of the Crazyflie](https://github.com/bitcraze/crazyflie-firmware/tree/master/examples/app_appchannel_test). The python class for the app channel is in the cflib here: https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/crazyflie/appchannel.py

Just mind that if you have any questions about the ecosystem, it is best to start a discussion in https://discussions.bitcraze.io/, since that is the main place for support. We use issues to indicate pieces of 'work' and changes necessary to the firmware/software

gemenerik, 2025-01-23:
Closing since this is not really an issue. Feel free to open a Discussions thread

closed 2025-01-23 by gemenerik