bitcraze/aideck-gap8-examples #145: Poor Classification Performance on Crazyflie AI-deck

luigifeola, 2024-09-05:
Following the steps in the [Classification Demo](https://www.bitcraze.io/documentation/repository/aideck-gap8-examples/master/ai-examples/classification-demo/), I was able to fine-tune the model with my custom dataset (with an higher number of images respect to the provided training data) as explained in the guide. I tested the model on my laptop using new images, and while it’s not perfectly accurate, it works most of the time and it is able to distinguish between two classes: `cap` and `background`. However, when I deploy the model on the AI-deck following the provided instructions, the network mostly detects only the background class.

Does anyone have suggestions on why this might be happening and how I can debug or resolve the issue?

gemenerik, 2024-09-06:
A great starting point [is to use `nntool`](https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/README.md#nntool-execution), as it allows you to feed inputs directly into your model and inspect what the network outputs. You can start debugging there.

Additionally I’ll give some pointers to help you troubleshoot, but the actual implementation and training fall under the user’s responsibility of course.

A good first step is to verify that the network performs well on your computer. This helps ensure that the model is properly trained and hasn’t fallen into issues like overfitting or poor generalization. For example, if your dataset is heavily skewed towards one class, such as mostly background images, the model might have learned to always predict background. This could result in the network getting stuck in a local minimum during training, which would explain the poor performance in certain cases.

Another important factor to consider is quantization. By default, the GAP8 deploys a quantized version of the network (quantized in TensorFlow), which can introduce performance drops due to reduced precision. Please check whether the quantized version of the network performs well on your computer.

You should also ensure that the dataset used for training closely matches the conditions of the deployed environment. Deep learning models can be highly sensitive to environmental variations. If the deployment setting differs significantly from the training environment, this could be affecting the network’s accuracy. Similarly, verify that the images fed to the network are in the expected format, including dimensions, color channels, and any preprocessing steps like normalization or scaling. A mismatch in format can severely impact the network’s ability to make accurate predictions.

luigifeola, 2024-09-06:
Thank you for your response.

- I will look into using nntool to inspect the network outputs and further debug the issue.
- The network performs well on my computer, it works with both Lite and quantized models. The dataset is not heavily skewed toward the cap class versus the background class, and while the accuracy isn’t very high, the model is still able to distinguish between the two classes as I mentioned in the initial comment.
- I have collected a dataset in two different environments and ensured that the deployment environment matches the conditions of the training images, including lighting and other relevant factors.

gemenerik, 2024-09-09:
Are you using the color or grayscale camera?
Curious to hear what comes out of your nntool inspection

luigifeola, 2024-09-10:
I am using the grayscale camera. 
In the meantime this is what I get when I run `tools/build/make-example examples/ai/classification clean model build image`. Since there are too many rows I will attach a file with the output.
[nntool.txt](https://github.com/user-attachments/files/16946142/nntool.txt)

gemenerik, 2024-09-11:
I meant to suggest manually using `nntool` to debug the model by loading tensors into the inputs and reviewing the outputs. Check out the [readme](https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/README.md#nntool-execution).

```c
classification.c:68:4: warning: pointer targets in passing argument 1 of 'classificationCNN' differ in signedness [-Wpointer-sign]
   (cameraBuffer, Output_1);
    ^~~~~~~~~~~~
In file included from classification.c:32:0:
/tmp/bitcraze/aideck-gap8-examples/examples/ai/classification/BUILD_MODEL_SQ8BIT/classificationKernels.h:535:12: note: expected 'signed char * restrict' but argument is of type 'unsigned char *'
```

may be problematic

luigifeola, 2024-09-19:
- Regarding the following warning:

```
classification.c:68:4: warning: pointer targets in passing argument 1 of 'classificationCNN' differ in signedness [-Wpointer-sign]
   (cameraBuffer, Output_1);
    ^~~~~~~~~~~~
In file included from classification.c:32:0:
/tmp/bitcraze/aideck-gap8-examples/examples/ai/classification/BUILD_MODEL_SQ8BIT/classificationKernels.h:535:12: note: expected 'signed char * restrict' but argument is of type 'unsigned char *'
```
What could I do to address this? As you probably know, `classificationKernels.h` is automatically generated in `BUILD_MODEL_SQ8BIT` folder. 

- I attempted to debug the model using `nntool`, as you suggested. I took inspiration from [nntool_script_q](https://github.com/bitcraze/aideck-gap8-examples/blob/master/examples/ai/classification/model/nntool_script_q). 
There is attached `output.md` with the terminal output, and maybe could be also helpful the `save_state` output reported in `classification_q.json`. 

- The link you provide me is really scarne in debug details, is anything more I can do to debug?

Thanks

[output.md](https://github.com/user-attachments/files/17062526/output.md)
[classification_q.json](https://github.com/user-attachments/files/17062522/classification_q.json)

gemenerik, 2024-09-23:
> What could I do to address this? As you probably know, `classificationKernels.h` is automatically generated in `BUILD_MODEL_SQ8BIT` folder.

You can convert your cameraBuffer to the expected type.

> ... is anything more I can do to debug?

I would try to get the model working as expected in `nntool` first. So I would indeed suggesting feeding it images and seeing if you can get your expected output. You can see how it was done in [the script you linked](https://github.com/bitcraze/aideck-gap8-examples/blob/master/examples/ai/classification/model/nntool_script_q#L18-L24).

luigifeola, 2024-10-03:
Which version of [docker-aideck](https://github.com/bitcraze/docker-aideck) are you using?

gemenerik, 2024-10-04:
[4.22.0.1](https://hub.docker.com/layers/bitcraze/aideck/4.22.0.1/images/sha256-038197df9cb86ccf8e6649e93dd0cf23781830e136288523983768918851633e?context=explore)

luigifeola, 2024-10-08:
> I meant to suggest manually using `nntool` to debug the model by loading tensors into the inputs and reviewing the outputs. Check out the [...]

I am using the same version. Do you also get the same warning when using the `nntool`?

gemenerik, 2024-10-09:
I am getting the same, try adding `imageformat input_1 bw8 offset_int8` after the `adjust` command in your `nntool_script` file.

luigifeola, 2024-10-17:
Hi @gemenerik, thank you for your help. The warning has been resolved, but now, regardless of the input, the network only outputs the custom class (in my case, the cap class). I'll continue investigating on my own, but any further assistance would be greatly appreciated.

gemenerik, 2026-10-19:
The example relies on autotiler which is no longer available. Closing as not planned but happy to give insights here should anyone seek it. Closest to an end-to-end example nowadays would be to train [PULP-Frontnet](https://github.com/idsia-robotics/pulp-frontnet) yourself and use it instead of the provided net in [the associated example](https://www.bitcraze.io/documentation/repository/aideck-gap8-examples/2026.10.2/examples/nanocockpit/pulp-frontnet/).

closed 2026-10-19 by gemenerik