bitcraze/aideck-gap8-examples #149: The wifi stream only works at the beginning, after disconnecting and reconnecting the code doesn't run. The code only takes a few images and then gets stuck.

Viteri-Z, 2025-05-08:
This is the code I changed in opencv-viewer.py, the raw image part will always get stuck, then the force quit and wait options appear, but the image will not continue to shoot, if you re-run it at this point it will get stuck at Connecting to socket on 192.168.4.1:5000... This is where the code will only work when I turn crazyflie on and connect it to wifi.Please help me to solve this problem！

import argparse
import time
import socket
import os
import struct
import numpy as np

# Args for setting IP/port of AI-deck. Default settings are for when
# AI-deck is in AP mode.
parser = argparse.ArgumentParser(description='Connect to AI-deck JPEG streamer example')
parser.add_argument("-n",  default="192.168.4.1", metavar="ip", help="AI-deck IP")
parser.add_argument("-p", type=int, default='5000', metavar="port", help="AI-deck port")
parser.add_argument('--save', action='store_true', help="Save streamed images")
args = parser.parse_args()

deck_port = args.p
deck_ip = args.n

print("Connecting to socket on {}:{}...".format(deck_ip, deck_port))
client_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
client_socket.connect((deck_ip, deck_port))
print("Socket connected")

imgdata = None
data_buffer = bytearray()

def rx_bytes(size):
    data = bytearray()
    while len(data) < size:
        data.extend(client_socket.recv(size-len(data)))
    return data

import cv2

start = time.time()
count = 0

while True:
    # First get the info
    packetInfoRaw = rx_bytes(4)
    [length, routing, function] = struct.unpack('<HBB', packetInfoRaw)

    imgHeader = rx_bytes(length - 2)
    [magic, width, height, depth, format, size] = struct.unpack('<BHHBBI', imgHeader)

    if magic == 0xBC:
        # Now we start rx the image, this will be split up in packages of some size
        imgStream = bytearray()

        while len(imgStream) < size:
            packetInfoRaw = rx_bytes(4)
            [length, dst, src] = struct.unpack('<HBB', packetInfoRaw)
            chunk = rx_bytes(length - 2)
            imgStream.extend(chunk)
     
        count = count + 1
        meanTimePerImage = (time.time()-start) / count
        print(f"Image {count}: {meanTimePerImage:.6f}s ({1/meanTimePerImage:.2f} FPS)")

        if format == 0:
            # Process raw Bayer image
            bayer_img = np.frombuffer(imgStream, dtype=np.uint8)
            bayer_img.shape = (height, width)
            
            # Display raw Bayer image
            cv2.imshow('Raw', bayer_img)
            
            # Save raw image if requested
            if args.save:
                if not os.path.exists("stream_out/raw"):
                    os.makedirs("stream_out/raw")
                cv2.imwrite(f"stream_out/raw/img_{count:06d}.png", bayer_img)
            
            cv2.waitKey(1)
        else:
            # Process JPEG image
            with open("img.jpeg", "wb") as f:
                f.write(imgStream)
            nparr = np.frombuffer(imgStream, np.uint8)
            decoded = cv2.imdecode(nparr, cv2.IMREAD_GRAYSCALE)  # Convert to grayscale
            cv2.imshow('JPEG (Grayscale)', decoded)
            cv2.waitKey(1)

evoggy, 2025-05-21:
Hi! Have you updated to the latest ESP32-S3 firmware? There's a fix there for reconnecting.

Viteri-Z, 2025-05-22:
I used the following method, is this method no longer available?
Flash using the cfloader
A binary can be flashed to the ESP via a Crazyradio using the cfloader

Note: replace the radio address with the appropriate value for your Crazyflie

cfloader flash build/aideck_esp.bin deck-bcAI:esp-fw -w radio://0/30/2M